kvasir-io/Kvasir #4: Discuss arguement order in MPL::Find

porkybrain, 2015-02-10:
the order of the arguements in MPL::Find seems odd, we should discuss it

bjoekeldude, 2015-02-10:
im afraid to tell you but i dont see anything odd in the arguments? do you mean MPL::Find oder MPL::Detail::Find? but eitherway it doesnt seem odd to me. 
btw why did you create the namespace Detail twice?

porkybrain, 2015-02-10:
I use MPL::Less in the second Detail namespace so MPL::Less must be defined above it, in less I use the InternalUseOnly type which needs to be declaired before Less and be in Detail.

porkybrain, 2015-02-10:
In the STL for example there seems to be a convention of algorithm functions (like find) taking the container as the first argument and other parameters like a predicate or a value after the container. I don't know of any other library which takes the predicate first and the container second. Therefore I think we should switch them around. And follow the convention of container first and then other arguments for all algorithms

closed 2015-02-12 by porkybrain