kvasir-io/Kvasir #75: many chip headers not compiling

johndevereux, 2016-05-29:
The example header Chip/MKL27Z4.hpp compiles OK. But most of the others I tried do not, producing many errors. This was using gcc-arm-none-eabi-5_3-2016q1 from the GCC ARM Embedded project.

The errors vary with the chip but for example
main.cpp:
#include "Chip/STM32F30x.hpp"
...

arm-none-eabi-g++ -o build/example-main.o -c -Og -mcpu=cortex-m4 -mthumb -std=gnu++11 -Ibuild/.:./lib -I.:./lib -I/home/jd/output/Kvasir/Lib main.cpp
In file included from /home/jd/output/Kvasir/Lib/Chip/STM32F30x.hpp:2:0,
                 from main.cpp:2:
/home/jd/output/Kvasir/Lib/Chip/Unknown/STMicro/STM32F30x/GPIOA.hpp:8:15: error: too many decimal points in number
               0..15)
               ^
/home/jd/output/Kvasir/Lib/Chip/Unknown/STMicro/STM32F30x/GPIOA.hpp:11:15: error: too many decimal points in number
               0..15)
               ^
(continues for many screenfuls)

porkybrain, 2016-05-30:
Oops sorry, the code gen algorithm has been fixed but it looks like we have not re generated everything yet.

porkybrain, 2016-05-31:
Hope it works for you now, just regenerated all the chip files based on updated cmsis-svd and algorithm

johndevereux, 2016-06-02:
It is fixed for me - it compiles fine now.
I see you fixed the "\" path separators too (which I should have mentioned).
Thanks!

closed 2016-06-02 by porkybrain