kvasir-io/Kvasir #81: apply, write etc missing on STM32F072x

Sickeroni, 2016-11-25:
apply, write set etc don't work. 
works on MKL27Z4 trough generated "Chip/MKL27Z4.hpp" thats include generated "Io.hpp" in that includes "Register/Register.hpp". Io.hpp is missing on STM32F072x and therefore Registers.hpp. 

helperscript must be modified for this.

odinthenerd, 2016-11-29:
yes this is a problem, I think its part of a bigger question: "how best to make a split header implementation?" the way we are doing it now is not strictly split header because we include different named headers. It would be good if someone could, at the command line, decide which target to build.

odinthenerd, 2016-11-29:
work around for now it to just include Register.hpp

odinthenerd, 2016-11-29:
We should put the "include all" files in the same folder as the rest of the chip files and call it all.hpp, then it should include Register.hpp too.

odinthenerd, 2016-11-29:
Chip should not be a sub folder of lib because lib is an include directory and this confuses intelisense because it walks all sub folders