kvasir-io/Kvasir #82: wrong apply in seam with sets in list-constexpr

Sickeroni, 2016-11-26:
# if saving list in constexpr & using more than one set() than seamwrite is only showing one set();
 
## testcases: 
### fails with 
`0x400@0x48000014`
constexpr auto s = set(Kvasir::GpiobOdr::odr8),set(Kvasir::GpioaOdr::odr8),set(Kvasir::GpioaOdr::odr10));
        apply(s);

### works with
`0x100@48000414`
`0x500@48000014`

constexpr auto s = (set(Kvasir::GpiobOdr::odr8, Kvasir::GpioaOdr::odr8, Kvasir::GpioaOdr::odr10));
apply(s);

apply(set(Kvasir::GpiobOdr::odr8,Kvasir::GpioaOdr::odr8, Kvasir::GpioaOdr::odr10));

apply(set(Kvasir::GpiobOdr::odr8),set(Kvasir::GpioaOdr::odr8),set(Kvasir::GpioaOdr::odr10));

odinthenerd, 2016-11-29:
if you want to save as a list you have to write `list(set(...),set(...))`

Sickeroni, 2016-11-29:
wouldn't it be an improvment to change this?
i mean
> Making it easy to do the right thing, and hard to do the wrong thing

and i thing i did the mistake more than one time and other people will do this then too

odinthenerd, 2016-11-29:
yes, we could implement this by overloading the , operator, but that is widely considered evil for good reason. So there are conflicting goals and I think we are doing the least evil thing.

Sickeroni, 2016-11-29:
closing through default comma operator behaviour:
Built-in comma operator
In a comma expression E1, E2, the expression E1 is evaluated, its result is discarded, and its side effects are completed before evaluation of the expression E2 begins (note that a user-defined operator, cannot guarantee sequencing) (until C++17).
source: http://en.cppreference.com/w/cpp/language/operator_other

closed 2016-11-29 by Sickeroni