kvasir-io/Kvasir #83: Ignore of Mask in Register::Action

Sickeroni, 2016-11-27:
Mask is ignored on writeactions. Means wrong bits could be set.
tested only with Register/Seam.hpp, not on real hardwarecompile

testcode: 
apply(Kvasir::Register::Action<Kvasir::Register::FieldLocation<Kvasir::Register::Address<0x13371337,0x00000000>,(0b01<<30)>,Kvasir::Register::WriteLiteralAction<(0b11<<30)>>{});
      apply(Kvasir::Register::Action<Kvasir::Register::FieldLocation<Kvasir::Register::Address<0x13381338,0x00000000>,(0b11<<30)>,Kvasir::Register::WriteLiteralAction<(0b11<<30)>>{});

odinthenerd, 2016-11-29:
I think the seam just dumps inputs... not really sure if this should be changed because it allows you to see the errors if we don't mask it off. So I think this is not a bug its a feature.

Sickeroni, 2016-11-29:
i my opinion its less a feature.
we use seam to unittest stuff
if it compiles there but not for the target its maybe suboptimal. 
i use seam on my machine to test all writes etc. It would extremly help me to find bugs if everything that exec is doing is done on my machine. 
i could just crosscompile it and the static assert would assert, but at that point bugs should be already removed.

odinthenerd, 2016-11-29:
well in the case of a runtime write the unit test would want to know that its overflowing the field and we can't know at compile time if it could. So we should preserve the overflowed data for the unit test. if there is an assert it should be in the unit test. In hardware we mask off the offending bits to stop corruption but do not report an error. 

So the unit test is more stringent than the HW. This is good. We just need to add an assert to the unit test.

closed 2016-11-29 by Sickeroni