kvasir-io/Kvasir #84: Cleanup in ChipFromSvdGenerator

jacquelinekay, 2016-11-28:
I'm not the most Pythonic Pythonista who ever Pythoned but I have a few ideas for cleaning up the chip file generator.

It may make sense to use [EmPy](http://www.alcyone.com/pyos/empy/) templates instead of concatenating formatted strings representing C++ code. It may make the process a bit more readable, since you could have a template file that looks a bit more like the resulting C++ code with placeholders that get filled in from the parsed xml. I have some experience with EmPy, so I could do this if others think it would be useful. 

I couldn't find tests in place for results of the generator; if the chip generator gets rewritten it would be good to have those in place beforehand, to make sure the rewrite preserves functionality. Are there IP issues with adding example CMSIS SVD files to this repo for testing?

CvRXX, 2016-11-29:
I looked at a couple of svd files and most of them have licenses that allow distribution as long as you retain the license notice. So I think you could include these files as an example if you check the license first. 

However I'm not sure if it is a good idea to put the files in this repo because they are subject to change. Maybe we could make a repo with all the svd files.

odinthenerd, 2016-11-29:
https://github.com/posborne/cmsis-svd has all the SVD files. Probably good to keep them there since there are many people using that. @kblomqvist also made a SVD parser which actually, in retrospect, looks more inline with my views on architecture, maybe there is something to be learned there.

These guys https://github.com/hackndev/zinc are also using the SVD parser for rust stuff, if we rewrite in such a way that we factor out the actual codegen templates then it could also benefit that community. 

Essentially we need a property tree and a somewhat smart `getProperty` function which takes a key (or path) and returns a value. This `getProperty` function should also back track up the tree looking for default values defined in parents. Everything below that should be use case agnostic.

On top of that layer we walk the sub keys of "peripherals" creating a file for each then walk registers and fields.

kblomqvist, 2017-01-09:
Hi. I have been working with my tool lately by refactoring its SVD parser and adding the missing elements. If there's something you would like to have there please let me know. The parser can be used standalone too, in case you just need the parser and not the templating engine.

```python
from yasha.parsers.svd import SvdFile

with open('nrf51.svd', 'r') as file:
	svd = SvdFile(file)
	svd.parse()
	print(svd.peripherals[0])               # The first found peripheral
	print(svd.peripherals[0].registers[0])  # The first register in the first peripheral
	print(svd.peripherals_dict.keys())      # The names of all peripherals
```

Installation: `pip install yasha`