kvasir-io/Kvasir #97: Should we rename namespaces to lowercase?

odinthenerd, 2016-12-05:
In the interest of breaking things now rather than later it seems people don't like my style ;) basically types being UpperCamelCase and values and functions being lowerCamelCase is ok by most people but that fact that namespaces are also upper camel case bugs a lot of people. So we should probably change it. 

The one breach in this style is the nested type in Metafunction<T>::type. The problem is that TMP convention is to use lower case type but it is a type so according to this style guide it should be upper case. I think we can get away with this breach in our style guide and use lower case though because the user should never see this, its just library internals.

TorstenRobitzki, 2016-12-17:
As I wrote a lot of ruby code, I could live with your style, but for consistency with existing C++ code, I would prefer the usual C++ style (all lower, underscores, template parameters in camle case etc.)

TorstenRobitzki, 2016-12-18:
In addition, I would suggest to put all headers under one "kvasir" directory just to reduced the risk that a path or file name clashes with other parts of a project. Using lower case file and path names would be more consistent too.