kvasir-io/Kvasir #103: Kvasir licencing and copyright

CvRXX, 2016-12-13:
I'm writing the contribution file for Kvasir at the moment and I'm doubting how to handle copyright. 

The whole Kvasir name space will be licenced under the apache. @odinthenerd and I were thinking on putting all the contributes in the readme for each repo in de the order of there first commit. This would work except for the copyright. Who is going to have copyright in the license of each file? Do we refer to the readme?

Would love your feedback!

@odinthenerd @Sickeroni @chieltbest

odinthenerd, 2016-12-13:
boost license is probably better than apache.  Its actually quite a complex question, most for profit companies have a serious interest in defending themselves against IP lawsuits and if we don't present them with a good license and assurance that all the IP is ours to license they will not be able to use it. 
This may sound stringent to the open source community but in the embedded domain the vast majority of products are by for profit companies and to ignore their needs is essentially isolating ourselves. Boost for example is not usable by some companies because it poses a risk in the eyes of their legal departments.

We have been playing with the idea of founding a "Kvasir foundation". This would be more of an"mbed" model, every contributor would have to sign a contributor agreement and the rights are pretty centralized (most other big organizations are similar, [see Qt for example](https://www.qt.io/contributionagreement/)). 

This would also allow the Kvasir e.v. (or whatever legal form it assumes) to accept donations and sell support in order to fund further development. 

I'm open to input on this, the fundamental vision is to impact the way this industry develops firmware, I don't really have any preference how we go about it but this is my opinion on how we can maximize effect. Non negotiable is that kvasir should be free for everyone, only then will we truly have a positive impact.

CvRXX, 2016-12-14:
In my mind the advantage of a a foundation is that a foundation serves a certain purpose. This means that donations will go to that purpose and not some company maintaining Kvasir. I think it will make sponsorships and donations easier.

But the question for now remains. Are we going to stick to the boost license even if it prevents some companies from using Kvasir. Does anyone have experience with a good license that is allowed by (almost) all companies?

odinthenerd, 2016-12-14:
Its not so much about the license it self, actually I think its good to use a common license like boost because people already know what they are getting. The problem with boost is that people are not 100% sure that the lib author is actually the original author of the code in question. 

As of now I am 100% sure that kvasir is 100% origional development, I wrote most of it and know every other major contributor. If we look at Qt they don't seem to have the same problem as boost, most of the Qt code came from the Qt foundation. Thats my train of thought. As of now Its probably good to get some kind of agreement stating that people agree to give up their rights to their contributions to an eventually created Kvasir foundation who's mission is to improve the embedded development experience and provide a free framework aiming at becoming industry standard.