kvasir-io/Kvasir #108: register offset addition results in >32 bits and -Wc++11-narrowing

jacquelinekay, 2017-01-31:
Some registers sum the base address and register offset to a result that requires more than 32 bits/4 bytes, which is invalid for a 32-bit address space. For example:

https://github.com/kvasir-io/Kvasir/blob/master/Lib/Chip/Unknown/STMicro/STM32L15xxE/GPIOE.hpp#L6

This results in -Wc++11-narrowing when trying to compile code that includes the header, since the type of all register field values is `unsigned`.

This is the section from the SVD that results in the addition:
https://github.com/posborne/cmsis-svd/blob/master/data/STMicro/STM32L15xxE.svd#L4172-L4183

As you can see, 0x40021000 + 0xFFFFF800 = 0x140020800

I came across this while compiling tests for the new generator. I am not sure how to fix it; it seems like there's either a mistake in the SVD file or we're missing a step in how to sum the offset and the base address. Do we need to apply a different operator? Do we need to wrap around? Am I missing something with my assumption about the size of the address space?

CvRXX, 2017-02-09:
@odinthenerd Do you know how this works?

odinthenerd, 2017-02-14:
check out http://www.st.com/content/ccc/resource/technical/document/reference_manual/cc/f9/93/b2/f0/82/42/57/CD00240193.pdf/files/CD00240193.pdf/jcr:content/translations/en.CD00240193.pdf page 47. The SVD is screwed up.