kvasir-io/Kvasir #118: Run doxygen and publish to gh-pages branch

LegalizeAdulthood, 2017-12-18:
One way to encourage use of Kvasir is to have documentation as readily accessible as possible.

The wiki seems a little stale and doesn't cover all the information you need.

There is a Doxyfile, but I haven't attempted to run doxygen to see what is produced.  If this is the intended way to document the API, then we should regularly run doxygen and push the results to the [github pages](https://pages.github.com/).

odinthenerd, 2017-12-21:
doxygen is a bit hard sometimes with heavily templated code. We have been using standardese on some other kvasir projects. lack of documentation is a real problem I agree.

LegalizeAdulthood, 2017-12-21:
Have you looked at [doxypress](http://www.copperspice.com/documentation-doxypress.html)?  It uses Clang as the front end, so should have better luck at parsing modern C++.

LegalizeAdulthood, 2017-12-23:
I'm trying doxypress right now... it's been grinding for several hours, LOL.  I think what I need to do is configure it to ignore the 770 MB of generated chip files.

odinthenerd, 2017-12-24:
yes the sheer volume of generated files needs to change