kvasir-io/Kvasir #119: Writing all bits of a register doesn't result in a blind write

LegalizeAdulthood, 2017-12-21:
Having gotten my basic GBA prototype working, I attempted to reproduce the [first tonc demo](http://coranac.com/tonc/text/first.htm) and compared the assembly dump of my .o to the one produced by that first code example.  Their code results in a blind write to the DISPCNT register, but my Kvasir code results in a read-modify-write to the register.  At first I thought this was because I wasn't writing all fields of the register, but even when I did that, I got a read-modify-write implementation.

I thought kvasir would optimize writing to all bits in the register as a blind write?

Perhaps I am not defining my DISPCNT register correctly (see [GBA.hpp](https://github.com/LegalizeAdulthood/Kvasir/blob/gba/Lib/Chip/Unknown/Nintendo/GBA.hpp)), or I am doing something wrong in using my register definitions?

My first assumption is user error on my part :)

Is there a shortcut for saying "clear all bits/fields in the register except those explicitly given a value"?

odinthenerd, 2017-12-24:
internally kvasir should be oring all the bitmasks of modified fields with the "has default value" bitmask and if that ends up as 0xFFFFFFFF then it should turn it into a blind write. I'll have a look and get back to you.

odinthenerd, 2017-12-24:
it might be that I hard coded it to look for 0xFFFFFFFF rather than the actual width of the register, I see this is a 16 bit register

LegalizeAdulthood, 2018-01-02:
Hrm.  I see that the mask helper function I was using is generating 32-bit masks.  I will adjust my field definitions and see if that gets me to a blind write.