kvasir-io/Kvasir #121: What's the state of this lib?

jankap, 2018-01-09:
Hello all,
this is an awesome project and a great idea of TMP. 
Before I'm going to dig deeper into detail I'd like to ask if you would say that this lib can be used in all day development already or if it's "only" a proof of concept with many gaps to be closed? It looks like there is a lot of work done in the mpl repo (https://github.com/kvasir-io/mpl), will this replace https://github.com/kvasir-io/Kvasir/tree/master/Lib/Mpl for example?

Thank you very much, keep up the great work!
Jan

odinthenerd, 2018-01-15:
This is a proof of concept, it has been used in production code but only by teams containing some of the authors. We ran into problem of the mpl part not compiling fast enough or crashing with complex programs. That should be solved now as the new mpl is like 30x faster and 20x more scalable. Most of the other parts are waiting for mpl to freeze before they go forward. There will be a road map discussion at embo.io regarding the future of this effort.

jankap, 2018-01-15:
Great, thank you very much for these information. Trying very hard to join the emBO conference but probably only one day will be approved ;) Is that discussion planned for a certain day? Can't find it in the schedule.
Thanks

odinthenerd, 2018-01-16:
day1 is training, day2 are presentations and the first half of that and day 3 is more of an experts discussion in a small group