kvasir-io/Kvasir #123: Portability to other architectures besides ARM?

DarthRubik, 2018-05-21:
Out of curiosity, how easy is it to generate a configuration file by hand for another architecture of micro controller/is that even possible with the current library?

I am just trying to understand more about the library.....from what I gather this is not possible?

Thanks

Sickeroni, 2018-05-21:
Its possible and already planned. Generators are just not programmed for other architectures. If you generate some manually, it should work. 
If you want to learn more about the library try the newer repos. 
This Repoo is more or less a Proof of Concept. the other Repos (gpio, bit,chip, toolchain, mpl etc) are modern but not fully working implementations of a newer Kvasir. Every Repo is for a specific part of the environment. 
For generators try kvasir-io/HardwareDescriptionFileGenerator
its not complete but a good startingpoint!

Sickeroni, 2018-05-21:
an interesting extension could be 
https://github.com/modm-io/modm-devices
its not kvasir, but this subproject of modm is catching and formatting data from stm (arm-processors) and atmel (AVRs & ARMs)

DarthRubik, 2018-05-21:
Ok, thanks for the info.......I will certainly look into that