kvasir-io/Kvasir #126: What is the state of Kvasir?

messi1, 2019-06-12:
Hi everyone.
thanks for the good tool collection.
I would like to know if there is still someone working on the lib?
The last commit was 2yr ago.

Are there any future plans like updating the doc, examples, ....

Thanks a lot 

Best regards

Juergen

liarokapisv, 2019-11-06:
@messi1 It appears that the project is abandoned until more resources are available or until a larger community effort is put forth. You may want to look at [Modm](https://github.com/modm-io/modm) as a more active alternative.

odinthenerd, 2019-11-11:
hey thanks @liarokapisv for making this pop back up un my messages ;) yes I do not have time for this project in the near future. if anyone wants to take over I would be open to that. Modm is far better maintained, would suggest that. Also, a lot of stuff coming out of the rust embedded community is along with the philosophy of this effort.

messi1, 2019-11-11:
@liarokapisv Thanks a lot for the hint. It was already on my list but also my time is unfortunately very limited.

eyalroz, 2021-02-18:
@odinthenerd : 

I've found this repository after watching your 2017 code::dive presentation. So, I was expecting to find a metaprogramming library and a better alternative to std::tuple... instead, I find an abandoned "Boost for microcontrollers".

Can you please take the time to at least change the README.md to say whether any part of the project has been picked up / is still maintained elsewhere? Especially the parts discussed in your presentation? Links woul dbe quite useful.

Also, what is "Modm"?

odinthenerd, 2021-03-05:
you are probably looking for https://github.com/kvasir-io/mpl